parawanderer/OpenTagViewer #1: Query AirTags for user automatically?

rohitgeo, 2025-03-23:
Great project... thanks!

Why do we need the step to Manually Export AirTags from a Mac and import them in the app? Can't the list of user's AirTags be queried automatically through some API once the user signs in using their Apple ID? I'm assuming the info is there on Apple's servers and I'm able to fetch the list of AirTags on a new Mac using FindMy even though I originally set up the AirTags from another device (iPad).

parawanderer, 2025-03-24:
Hi @rohitgeo,

So what that step is doing is basically extracting private keys for being able to query the FindMy network from the Mac's cache. Because Macs have a very convenient location where they put a bunch of cached information. I absolutely agree that this is very contrived and would ideally be skipped.

Now the problem is that the [FindMy.py](https://github.com/malmeloo/FindMy.py) library (used for querying the FindMy network) that my project is built upon is 100% reverse engineering work, i.e. no actual support is provided for any of this by Apple itself. 

So while I am absolutely certain that there is a very pretty and easy to use API call available from Apple's side that fetches the data on the airtags available to the current user, this API call is intentionally hidden from anybody outside of official Apple devs.

<br>
I think it is possible to reverse engineer both this "fetch device data" (which requires a Mac/VM in my project)  and even the "register new AirTag with apple" (which requires an iOS device in my project). I don't think it has been done yet, though. And I'm not sure if anybody is currently trying to do this either.

<br>
<br>
Either way -- should you find a project that managed to fetch either data from Apple (available devices or registering a new device) let me know and I will integrate it in this project! Until then, I would suggest <a href="https://github.com/malmeloo/FindMy.py/issues">opening a feature request with the FindMy.py library for these to be added</a>, so that somebody might feel there's enough demand to pick up those investigations 😛 

When/if this is achieved publicly by anybody, I'm happy to integrate it in this project.

parawanderer, 2025-08-06:
Closing this. Would definitely be happy to implement automatic behaviour once a way to do this is discovered, but for now practically not possible unless somebody figures out how to do it without the cache files!

closed 2025-08-06 by parawanderer