parawanderer/OpenTagViewer #2: Weired Behavior

daoudeddy, 2025-03-25:
I'm trying to setup this app on android, i can successfully manually export the plist from a qemu mac, logged with my 2FA Apple account, imported the zip the devices are shown but the location isn't updating.

[image]

parawanderer, 2025-03-25:
Hi, 

Are you able to manually reload them in the main map display and does that produce any error messages? Does it try to refresh automatically that you notice (e.g. when you go to another app and then come back to this app in ~a minute)?

You can also try to go to the history page for every tag, and go back in the history for the last 7 days (that's how long the Apple servers keep location data around for) and see if it matches to retrieve that for you, if you haven't tried doing that for those tags/days yet. If it does find data for previous days then that means the data retrieval part is (still) working. Furthermore, any time you go to the history page for "today", it will refetch the history from scratch for the last 24hrs (this is not the case for historical data older than 24hrs, this will be cached after one successful retrieval). 

I sometimes saw a similar behaviour when my tags were just sitting around in a place where no Apple devices were around to ping them and send the location updates off to Apple. And walking around in a more populated area with them resulted in the updates coming in. Note that this app does not itself send updates about your tags' location to Apple at this moment in time -- only actual Apple devices can do that atm. So all this app is doing atm is retrieving those submitted reports from Apple. 

There's also a pending item on the backlog for this app to enable scanning your local Bluetooth area for tags and report those in the UI if it can find them (in addition to the history it fetches from Apple). But unfortunately that's blocked for now, due to a bug in the FindMy.py library that this project relies on

parawanderer, 2025-03-26:
Hmm, I think it's worthwhile to check whether the app is throwing some errors for sure, so if you would like to do that that would be much appreciated. I honestly don't think whether it matters if they are official Apple-produced AirTags or not -- I understand that if they are compatible per Apple then they should essentially behave the same when it comes to location reporting to the Apple servers and Bluetooth as the Apple-produced ones.

The `.apk` on the downloads page isn't the debug build, it's the [release build](https://github.com/parawanderer/OpenTagViewer/blob/0c0be1a1540ff25152521d42683ddb09c22387fa/app/build.gradle.kts#L39) but from what I could see when I was testing this, you can still see the logcat logs in this release build. Just filter by the package `dev.wander.android.opentagviewer`. I think what you can't do is *attach to the process* for debugging. The errors should definitely be getting dumped in a few places that I removed visual error messages.

But otherwise I am wondering if you might not be getting affected by this issue: https://github.com/malmeloo/FindMy.py/issues/46#issuecomment-2223726526
Which is currently an open issue here (that's why I can't currently include this functionality in this app): https://github.com/malmeloo/FindMy.py/issues/90

In my case, my iPad will show the airtags nearby and update very frequently when I'm physically them with the iPad because at that point it's retrieving that data from checking the local Bluetooth rather than querying the Apple servers. In this case the history

Physfock, 2025-03-31:
Why does 'no last location known' appear here? I have the same issue. Only the devices that were added relatively recently are visible, while all device locations exist on iOS.

parawanderer, 2025-03-31:
@Physfock hi, so I think I can probably explain why this could happen. 

So as I mentioned earlier in this thread, Apple only keeps around the last 7 days worth of location history on their servers. 

Now keep in mind that device history updates only come available in the Apple FindMy servers when some **iOS** device detects the device locally by Bluetooth and then pushes the information to the Apple Servers. So no apple devices nearby that have Bluetooth on AND that can themselves connect to the internet = no updates.

And in the main app UI, what happens is that every 1 minute that the app is being directly viewed on your screen, it will run a task where it fetches the last 24 hours worth of location history from that moment backwards (when this happens you will see the spinner icon appear on the tags). So if a tag were to suddenly get a new location update after not having had any, it should also at that moment appear in the UI, alongside everything that already had history.

Note that I made the decision to not fetch the last 7 days worth of history but only the last 24 hrs, mostly based on the fact that I had some personal test cases where a single day had >1k datapoints, so fetching 7 days in one go every time seemed like it might be an awfully large and suspicious "ban-worthy" query. 

<br>
Sidenote: you can actually make the app retrieve the last 7 days manually, even for devices that are shown as "no last location known", by going to that device's settings page and going to the "Device History" tab from the "more options" drop-down (top right) on that page, and then just going back to each day of the last 7 days in history and having those get loaded successfully. This makes the app fetch the full day's worth of history and store it permanently locally.

<br><br>
Now you could technically identify devices that are **locally available** via Bluetooth, without going to the Apple servers. But due to that aforementioned bug in the FindMy.py library from my previous comment, this app isn't doing that at the moment. The fix for that is pending by that library's author or some other contributor (I personally don't have time to look into that atm, have other projects and stuff I need to work on).

I also don't believe there's a way in any public FindMy library to ask for just the "last update" **generally**, (i.e. regardless of time range), but it does seem a bit strange that such a functionality would just straight up not be implemented by Apple... So I'm wondering if this might just something that just hasn't been found/reverse-engineered yet? Surely that is what Apple uses in their UIs, because just storing one "last day" somewhere, regardless of how long they keep "history" around for, is quite cheap and trivial to do per device registered in their network. So I wouldn't buy a conclusion that this "just doesn't exist" on Apple's side

closed 2025-08-06 by parawanderer