parawanderer/OpenTagViewer #3: MacOS Sequoia (15)+ support

rohitgeo, 2025-03-26:
What options do I have when using MacOS Sequoia? The script to export the tags isn't working. Did they change the place where the tag data is stored?

parawanderer, 2025-03-26:
Yeah I think this comes down to this issue: https://github.com/malmeloo/FindMy.py/issues/31#issuecomment-2627204121

If anybody manages to figure out how to access that key, then we could test if the key still works and add support for it in the exporter script. 

I was basically too lazy to check that out since I don't run a Mac as my main OS, and just choosing to spin up my MacOS VM as Sonoma/14 instead of Sequoia/15 was easy enough.

parawanderer, 2025-04-13:
Hi @rohitgeo , somebody came up with a workaround to get the AirTag data on `MacOS 15+`. It's a bit involved such that I am not sure whether I should include this in my exporter app, requiring system restarts etc. But I included the approach as a possible option in the "[How To: Manually Export Airtags](https://github.com/parawanderer/OpenTagViewer/wiki/How-To:-Manually-Export-AirTags#process)" wiki page.

plabzkhn, 2026-02-05:
mac os 26.1 not working, any workaround?

parawanderer, 2026-08-15:
Hi, this is now fixed: https://github.com/parawanderer/OpenTagViewer/releases/tag/exporter-v1.1.0

The new iCloud route doesn't read macOS's files at all, so the OS version stops mattering.

closed 2026-08-15 by parawanderer