parawanderer/OpenTagViewer #7: Can you add Japanese?

horihori2013, 2025-04-26:
Not a problem, but I'll put it here.
I have a strings.xml file that I have translated.
URL:https://drive.google.com/file/d/1GyeTGlYv41D2EoXqzdLTA4sxsMbr4pE8/view?usp=sharing

horihori2013, 2025-04-26:
and can you do it for korean too?
I also have a strings.xml file that I have translated.
URL:https://drive.google.com/file/d/1bYBjyv0EMRCUN52GmWKH0--UAUpPjG5L/view?usp=sharing

parawanderer, 2025-04-27:
Hi @horihori2013 , thank you for the contribution 😄! I will try to review this & merge by EOD today

parawanderer, 2025-04-27:
Hi @horihori2013 , so I ran this through a translation check tool and it flagged some potential problems with the translations in the Korean file. (But of course no idea personally if those are valid or not 😅 ).

Could you check if the proposed corrections here make sense? If they don't make sense I will merge your proposed files as they are, without the changes: #8

Btw as a sidenote, I think it would be nice if you got added to the contributors list for the project for adding these, but I'd need a git commit from you for that, or alternatively I can stage the commit to some email you use. What that would do is list you on the home page of the project as a "Contributor". Is that something you'd be interested in? If yes, which email would I be able to use for that (same name/email as your commits on [this repo](https://github.com/horihori2013/BS-Browser) maybe)?

parawanderer, 2025-05-28:
Merged a version of both for now, which means that Japanese and Korean are available in the last release! Thanks again for the contribution! 🙏

closed 2025-05-28 by parawanderer