parawanderer/OpenTagViewer #12: No last location known

coat08, 2025-06-05:
So iv followed all instruction iv got the tags on the OpenTagViewer app but no last known location and the map is just google map.

Iv got an old iphone and I can see the trackers. am I missing something ? I'm even running my own anisette server. 

Thanks.

coat08, 2025-06-05:
Iv just noticed it says error importing new devices but all tags are there, it says to restart app and try again.

parawanderer, 2025-06-08:
I think this might be related to past issues some people have reported here with regards to some tags just not working. 

See: https://github.com/malmeloo/FindMy.py/issues/90

I'm not sure that there is a solution if you happen to run into those tags. I think I've seen some people unlink and relink their tags and that fix them maybe? But not sure. We're mostly all waiting on that issue above to be fixed

parawanderer, 2025-08-06:
Hi, I'd like to refer you to this new log dump feature, to be able to debug the issue better for your case:
https://github.com/parawanderer/OpenTagViewer/issues/19#issuecomment-3105164390