parawanderer/OpenTagViewer #13: Permission to access BeaconStore not granted

wes1993, 2025-06-08:
Hello,
First of all, thank you so much for your fantastic project! 😄

I’m following the instructions to use the macOS UI Export Wizard (as described [here](https://github.com/parawanderer/OpenTagViewer/wiki/How-To:-Export-AirTags-From-Mac#macos-export-wizard--recommended)), but I'm running into a problem: the password prompt never appears, and I get the error:

Permission to access "BeaconStore" not granted

I’d really appreciate any help you can offer.

Best regards,
Stefano

parawanderer, 2025-06-08:
Hi @wes1993, are you on MacOS <= 14 for this? Could you manually try running the command 

```bash
security find-generic-password -l 'BeaconStore' -w
```

In your terminal? I don't need the output here, just curious if you'd get an error.

wes1993, 2025-06-08:
Hello @parawanderer,
Thank you for your quick response! 😄 I'm running Ventura 13.3.1 inside a Proxmox VM (QEMU).

When I run:
```
security find-generic-password -l 'BeaconStore' -w
```
I get:
```
security: SecKeychainSearchCopyNext: The specified item could not be found in the keychain.
```
However, when I run:
```
security find-generic-password -l 'BeaconStore'
```
I can see all the values, including the gena blob.

Best Regards
Stefano

wes1993, 2025-06-08:
P.S. Manually supplying the gena value in the Python script allows decryption to work correctly. Therefore, I’d like to request a feature: if the automated retrieval fails, please fall back to allowing the user to enter the gena value manually using the command ```security find-generic-password -l 'BeaconStore' -g```.

parawanderer, 2025-06-08:
Oh wow. Well it being **empty** is definitely unexpected by my implementation. I wonder if that is a 13/Ventura problem? (I tested my setup on 14/Sonoma)

Honestly I'm confused why you get an empty string? On 15/Sequoia the help message for the security CLI tool shows:
```
$ security find-generic-password -h
Usage: find-generic-password [-h] [-a account] [-s service] [options...] [-g] [keychain...]
    -a  Match "account" string
    -c  Match "creator" (four-character code)
    -C  Match "type" (four-character code)
    -D  Match "kind" string
    -G  Match "value" string (generic attribute)
    -j  Match "comment" string
    -l  Match "label" string
    -s  Match "service" string
    -g  Display the password for the item found
    -w  Display only the password on stdout
If no keychains are specified to search, the default search list is used.
        Find a generic password item.
```

I'm also not completely sure that your previous error message is related to this empty string problem:

```
Permission to access "BeaconStore" not granted
```

This seems like it's skipping the password input popup somehow for you (that I got on Sonoma). 

I'd almost be tempted to suggest to run it with Admin permissions if you're not already? Not sure how to do this on MacOS personally, but I know my new work laptop has similar seeming all the time and unblocking Admin permissions fixes these.

parawanderer, 2025-06-12:
Hi @wes1993 , I think this would work as a solution. I probably am not going to be free to implement this fallback logic until towards the later end of the month but otherwise will circle back to this and use this approach. If you are free you are also absolutely welcome to implement this extension to the python decoder module/script -- but completely fair if you don't want to or have time for this.

parawanderer, 2025-07-16:
Apologies for the lateness but I will look into this request now

parawanderer, 2025-07-16:
Hi @wes1993 really curious if you'd be available to test this on your VM as it is now (as I don't have enough time to spin a new one up for this). I implemented the fallback behaviour in the current main branch. 

TL;DR just clone this repo and follow [this README](https://github.com/parawanderer/OpenTagViewer/tree/main/python#-export-wizard-development) to run `python wizard.py` and then see if it lets you proceed from there as expected.

If not, no problem at all -- I will just delay packaging up a new release until a bit later.

parawanderer, 2025-07-19:
Should be fixed now but the "gena" field doesn't seem to be populated on my VM. So this is there as a fallback for now for the standard approach:

https://github.com/parawanderer/OpenTagViewer/releases/tag/macos-exporter-v1.0.3

closed 2025-07-19 by parawanderer