parawanderer/OpenTagViewer #16: Feature Request: Allow users to switch different map providers

hker9527, 2025-06-14:
First, thank you for providing an easy way to track airtags on Android devices! (No need to carry an iPhone around, yay)

Is it possible to add different map provider's support, say OpenStreetMap? Google Map's data may not be accurate and up-to-date for some countries like China. Plus, it can be a more free (freedom) app.

parawanderer, 2025-06-14:
Glad you like it! 😄 

This seems like a good idea and from what I remember probably straightforward enough to do, so I'll keep it for the Todo list for when I get some time to go through some other issues here.

satmandu, 2025-07-23:
May I humbly suggest maplibre as a way to use OSM data?

https://github.com/maplibre/maplibre-native

satmandu, 2025-07-23:
Also: https://maplibre.org/news/2024-12-12-maplibre-android-vulkan/