parawanderer/OpenTagViewer #17: Feature request: add Play Sound button

YOZZOPANOZZO, 2025-06-17:
Hi!

First of all, thank you for the amazing work! It’s great to be able to track AirTags on Android finally.
The only feature I’ve been missing is the Play Sound button. Is it by any chance on the roadmap?

Thanks!

parawanderer, 2025-06-19:
Hi @YOZZOPANOZZO , glad you found this useful!

This is a request to the upstream from me actually, https://github.com/malmeloo/FindMy.py

Specifically see this issue: https://github.com/malmeloo/FindMy.py/issues/88

I'm mostly just concerned with the Android UI wrapper in this project but if this gets implemented per that issue then I am happy to add a button for it (pretty easy/low effort)

parawanderer, 2026-09-16:
Hi, thanks to @ubrt this feature is now available in [this](https://github.com/parawanderer/OpenTagViewer/releases/tag/android-app-v1.1.0) release!  🎉🎉🎉🎉

closed 2026-09-16 by parawanderer