parawanderer/OpenTagViewer #19: Error occurred while trying to import new devices

SomeAspy, 2025-06-29:
I used the MacOS 15 export method and created the zip from the wiki.
However, when I attempt to import the zip, I get the following:
```
Loading location data for 6 new imported devices ...
Error occurred while importing new devices. Try to restart the app and r...
```

Device: Pixel 8
Version 1.0.3
Android 16

I don't really have further details.

trek1, 2025-07-07:
Similar problems for me, but I used the MacOS Export Wizard running on Mac OS 13, so it doesn't appear to be the export method that's caused the issue.

Device:
Samsung A54 
Android 15

parawanderer, 2025-07-19:
Commit above isn't really fixing whatever issue you might be having, but should lower the effort needed to dump the error. It will let you export the last 500 internal app logs to a file. I suppose it is rather indiscriminate but it skips having to get an android dev setup or familiarising yourself with the tooling. Just strip out any irrelevant logs.

I'll create a build with this new log dump feature alongside a fix for #20 most likely tomorrow.

fraymackers, 2025-07-22:
> Ouch yeah this is a difficult one. From my side it's also very unspecific as to what might be happening when you get that error.
> 
> I will try to extend the app to be a bit better about being able to export debug logs.
> 
> The first idea I have in this case is whether the Anisette server you are using might be causing you this issue? I've seen similar error messages before when using a generic Anisette server and I set up my own to fix this (see [this guide for how to do that](https://github.com/parawanderer/OpenTagViewer/wiki/How-To:-Your-Own-Anisette-Server)).
> 
> You can of course also try to just connect to some other Anisette server and see if that works.
> 
> But if it's not that, I think I'll basically need to see the debug logs. I know it's a huge ask, but if anybody has free time on their hands, it would be hugely appreciated if they could get the logs from when that error occurs via `adb` (Android Debug Bridge). 🙏
> 
> But if not, give me a bit and I will add a "export debug logs" utility to this app...

Hi, you are right! I had this import bug and changed the default Anisette server with my own server and it solved the problem for me 👍

parawanderer, 2025-07-22:
Hi all, so the latest release of the Android App available here should now have the "Export Logs" feature available in it:

https://github.com/parawanderer/OpenTagViewer/releases

TL;DR you need to enable `Enable debug data` in the Settings page and once you do that the option to export debug logs becomes available from the main map vertical `...` dropdown.

satmandu, 2025-07-23:
@parawanderer Do those logs have any information that should be sanitized before sharing to help debug some of these issues?

SomeAspy, 2025-07-29:
Hi, Sorry I got busy.
I am using the public anisette server (ani.sidestore.io)
I'll get you the logs in a moment.

schrammtastisch, 2025-07-31:
I had the same issue but I can confirm that choosing another Anisette server from the list did help.

parawanderer, 2025-08-02:
Right so for this login thing I would highly recommend checking if using a different server helps  @SomeAspy . Not yet sure why this happens but I saw this `LOGIN_STATE` issue occur for the default server before, and switching the server out to my self-hosted server fixed it permanently. You might also want to just do the equivalence of "turn it off and back on again" with the default server and see if that fixes it. 

I'll try to investigate if it's some app state management issue when I have some time.

malmeloo, 2025-08-03:
Public anisette servers should not be used if you need a long-lived session (which this app needs), because they tend to wipe their state every so often, leading to exactly these kinds of issues. Using a private or self-hosted one fixes it, or wait for the next release of findmy.py for integrated local anisette generation (should be out in the next 1-2 weeks)

malmeloo, 2025-08-05:
I ended up taking that guy's emulator code and turned it into a library that's nicer to use. I also made some improvements to make it a lot more stable than his initial version, and although there is still a (very minor) memory leak, it's been mitigated by restarting the VM if it gets out of hand.

Regardless of the platform used we will still need some sort of VM to use the native libraries. They're not meant to be run on their own; the VM provides things such as the memory allocator and manually sets/reads registers that are necessary to use the correct functions.

Unicorn (the VM engine) uses qemu as backend, and I do believe qemu comes with some optimizations when running on the same architecture as the host. Whether those will be enabled and what their effect would be I cannot say, so my recommendation would be to just try it and see.

SomeAspy, 2025-08-06:
> [@SomeAspy](https://github.com/SomeAspy): Right so for this login thing I would highly recommend checking if using a different server helps.
> 
> Not yet sure why this happens but I saw this `LOGIN_STATE` issue occur for the default server before, and switching the server out to my self-hosted server fixed it permanently. You might also want to just do the equivalence of "turn it off and back on again" with the default server and see if that fixes it.
> 
> I'll try to investigate if it's some app state management issue when I have some time. I think it might be. Did you ever choose a different Anisette server from the server dropdown? Alternatively: what steps did you do when setting up the app as far as non-default choices go (if you remember)?

I do not remember very clearly, I'm not selfhosting so I don't think I would do anything non-standard.
I've never attempted changing the anisette server though.

parawanderer, 2025-09-09:
This is now also waiting on this to be implemented (mostly): This is now waiting on this to be implemented: https://github.com/parawanderer/OpenTagViewer/issues/30

parawanderer, 2026-09-16:
Hi, this should now be fixed in [the new 1.1.0 release](https://github.com/parawanderer/OpenTagViewer/releases/tag/android-app-v1.1.0), can you have a look? If it is ongoing in any new shape or form, please feel free to open a new bug issue.

closed 2026-09-16 by parawanderer