parawanderer/OpenTagViewer #24: Export Wizard on BigSur (11.7.1)

ThinkMaize, 2025-08-02:
Despite giving the application permission to run, I cannot get the Export Wizard to work. When I click on the application, it will indicate that the app requires access to my keystore, but as soon as I click OK, everything terminates and disappears. Is this application supported on BigSur?

[image]

ThinkMaize, 2025-08-04:
I tried to do the manual mode as you suggested, but ran into errors that way as well. It could be something simple that I couldn't figure out because I am just not very familiar with the Mac OS. I was never asked for permission to access my keystore like I was when I used the wizard. Here is a copy of the output in case that helps:

```
bash-3.2$ python airtag_decryptor.py 
Traceback (most recent call last):
  File "/Users/user1/my_projects/OpenTagViewer/python/main/airtag_decryptor.py", line 60, in get_key
    return extract_key(key_in_hex_format)
  File "/Users/user1/my_projects/OpenTagViewer/python/main/airtag_decryptor.py", line 70, in extract_key
    raise KeyStoreKeyNotFoundException("Key was empty!")
KeyStoreKeyNotFoundException: Key was empty!

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/Users/user1/my_projects/OpenTagViewer/python/main/airtag_decryptor.py", line 275, in <module>
    main()
  File "/Users/user1/my_projects/OpenTagViewer/python/main/airtag_decryptor.py", line 258, in main
    key: bytearray = get_key(KEYCHAIN_LABEL)
  File "/Users/user1/my_projects/OpenTagViewer/python/main/airtag_decryptor.py", line 63, in get_key
    raise KeyStoreKeyNotFoundException("Failed to retrieve keystore value") from e
KeyStoreKeyNotFoundException: Failed to retrieve keystore value
```

ThinkMaize, 2025-08-06:
It does now specify that the command with the -w flag returned empty, but I'm not sure that the updated version of the code is switching to the backup method because it is still stopping with that error. I am on Python 3.13.0 rather than 3.13.2, but I doubt that would make a difference. I can confirm the key from the previous post is 64 characters, excluding the leading 0x. If it would be helpful for troubleshooting, I can send it to you as a DM. Thanks for hiding the latter part.

```
bash-3.2$ python --version
Python 3.13.0
bash-3.2$ python airtag_decryptor.py 
Traceback (most recent call last):
  File "/Users/user/my_projects/ManualTag2/airtag_decryptor.py", line 76, in get_key
    raise KeyStoreKeyNotFoundException("security command output with flag -w returned empty")
KeyStoreKeyNotFoundException: security command output with flag -w returned empty

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/Users/user/my_projects/ManualTag2/airtag_decryptor.py", line 300, in <module>
    main()
    ~~~~^^
  File "/Users/user/my_projects/ManualTag2/airtag_decryptor.py", line 283, in main
    key: bytearray = get_key(KEYCHAIN_LABEL)
                     ~~~~~~~^^^^^^^^^^^^^^^^
  File "/Users/user/my_projects/ManualTag2/airtag_decryptor.py", line 81, in get_key
    raise KeyStoreKeyNotFoundException("Failed to retrieve keystore value") from e
KeyStoreKeyNotFoundException: Failed to retrieve keystore value
bash-3.2$ 
```

ThinkMaize, 2025-08-07:
This updated script executes without error. However, it seems to only generate half the filetree, with no OwnedBeacons folder. I created a zip with just these files and the yml I created, but the app wasn't able to import it. Maybe this will fix itself, though, once your wizard gets the latest code update?

Generated File Tree:

<generated output folder>/
├── BeaconNamingRecord/
│   ├── <UUID 1>/
│   │   └── <UUID 2>.plist
│   └── ...
└──

```
bash-3.2$ python airtag_decryptor.py
Trying to decrypt plist file at: /Users/user/Library/com.apple.icloud.searchpartyd/BeaconNamingRecord/003.......EAF.record...
Now trying to dump decrypted plist file to: /Users/user/plist_decrypt_output/BeaconNamingRecord/003......EAF.plist
Success!
Trying to decrypt plist file at: /Users/user/Library/com.apple.icloud.searchpartyd/BeaconNamingRecord/26B...664.record...
Now trying to dump decrypted plist file to: /Users/user/plist_decrypt_output/BeaconNamingRecord/26B...664.plist
Success!
Trying to decrypt plist file at: /Users/user/Library/com.apple.icloud.searchpartyd/BeaconNamingRecord/7E7...7AF.record...
Now trying to dump decrypted plist file to: /Users/user/plist_decrypt_output/BeaconNamingRecord/7E7...7AF.plist
Success!
Trying to decrypt plist file at: /Users/user/Library/com.apple.icloud.searchpartyd/BeaconNamingRecord/F77...0D2.record...
Now trying to dump decrypted plist file to: /Users/user/plist_decrypt_output/BeaconNamingRecord/F77...0D2.plist
Success!
Trying to decrypt plist file at: /Users/user/Library/com.apple.icloud.searchpartyd/BeaconNamingRecord/B81...BFB.record...
Now trying to dump decrypted plist file to: /Users/user/plist_decrypt_output/BeaconNamingRecord/B81...BFB.plist
Success!
Trying to decrypt plist file at: /Users/user/Library/com.apple.icloud.searchpartyd/BeaconNamingRecord/F24...6A2.record...
Now trying to dump decrypted plist file to: /Users/user/plist_decrypt_output/BeaconNamingRecord/F24...6A2.plist
Success!
```

parawanderer, 2025-08-07:
Let me dump you the structure of what an `OwnedBeacon` file looks like, @ThinkMaize . Maybe it changed between MacOS releases:

```xml
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>batteryLevel</key>
	<integer>1</integer>
	<key>cloudKitMetadata</key>
	<data>
		<!-- 
			Some LONG Base64 String.
			Think: ~2500 characters
		 -->
	</data>
	<key>identifier</key>
	<string>11111111-1111-1111-1111-111111111111</string>
	<key>isZeus</key>
	<false/>
	<key>model</key>
	<string></string>
	<key>pairingDate</key>
	<date>2025-02-27T20:03:32Z</date>
	<key>privateKey</key> <!-- IMPORTANT! -->
	<dict>
		<key>key</key>
		<dict>
			<key>data</key>
			<data>
			<!--
				Base64 String. ~100 ish characters
			-->
			</data>
		</dict>
	</dict>
	<key>productId</key>
	<integer>21760</integer>
	<key>publicKey</key> <!-- IMPORTANT -->
	<dict>
		<key>key</key>
		<dict>
			<key>data</key>
			<data>
			<!--
				Base64 String. ~80 ish characters
			-->
			</data>
		</dict>
	</dict>
	<key>secondarySharedSecret</key>
	<dict>
		<key>key</key>
		<dict>
			<key>data</key>
			<data>
			<!--
				Base64 String. ~40 ish characters
			-->
			</data>
		</dict>
	</dict>
	<key>sharedSecret</key>
	<dict>
		<key>key</key>
		<dict>
			<key>data</key>
			<data>
			<!--
				Base64 String. ~40 ish characters
			-->
			</data>
		</dict>
	</dict>
	<key>stableIdentifier</key>
	<array>
		<string>1111~#111111a11111aaaa~#ABCDEFGHIJKL</string>
	</array>
	<key>systemVersion</key>
	<string>2.0.73</string>
	<key>vendorId</key>
	<integer>76</integer>
</dict>
</plist>

```

Does yours match this? Because there is data validation in place to assert parts of this format

ThinkMaize, 2025-08-07:
The files within the MasterBeacons folder do match that exact format, so I repeated the process:
- Rename MasterBeacons > Owned Beacons
- Generate YML file using script provided
- Delete folders other than BeaconNamingRecord and OwnedBeacons
- Zip
- Import on Android phone (after clearing app storage and re-logging in)

And this time it imported all 5 of my items (4 Airtags, 1 phone). Interestingly, only 1 of the devices, an Airtag, was able to display a recent location, and the others just say "No last location known" with a little exclamation point in a triangle next to the name, but hopefully this will update once I give it more time.

I will follow up to let you know if the one device continues to update location and if the others start to show a location. I can certainly send you the zip if that would be helpful for updating the script.

parawanderer, 2025-08-08:
I think for the location update thing you do basically just have to wait for it to update. There are some issues with updates in general (see: #19 ) but @malmeloo is already working on a fix upstream from me https://github.com/malmeloo/FindMy.py/pull/160

One trick you can use to get some old locations to show up is to go to `the list of tags > settings drop-down > history` and then go back in history for max 7 days for every tag. This will force it to load and save that history locally (no data exists on apple servers past 7 days back). Do that for every tag separately. Then you will unfortunately have to close and reopen the app (dumb issue on my side that I need to fix) but afterwards it will show you the latest location from history until it manages to fetch new data.

I'll probably be looking to automate that backwards lookup better and fix the backwards history load soon-ish. 

But yeah regarding the import: if you could I would appreciate a zip of whatever crashed the app last time, because ideally that shouldn't happen outright!

I also have an idea now of how I can automate both the wizard and the script to deal with older MacOS generically with the rename, so I'll implement that fix in the exporters too.

parawanderer, 2025-08-15:
[This release](https://github.com/parawanderer/OpenTagViewer/releases/tag/macos-exporter-v1.0.4) should deal with the exporter app angle properly now. 

TL;DR both the standalone python script and the wizard will now try to extract `MasterBeacons` too and they will perform a rename step to make the format compatible with the app