parawanderer/OpenTagViewer #35: Connection is successful, but Apple always returns "0 reports"

lequan2909, 2025-09-24:
### Bug Description
After a complete and clean setup, the app authenticates successfully but is unable to display any
locations for my AirTags. The logs consistently show that the app successfully fetches reports from
Apple's servers, but the result is always "Got 0 reports".

This happens even after creating a fresh .zip export and ensuring the AirTag's location is visible and
up-to-date in the official Find My app on an iPhone.

### Key Findings
- There are NO authentication, DNS, or other connection errors in the logs.
- The self-hosted Anisette server (via Docker and Cloudflare Tunnel) is working correctly.
- The app successfully authenticates, fetches, but receives an empty data set from Apple.

This suggests the issue may be a change in the Apple API that the underlying findmy.py library is not
currently handling, rather than a setup or user error.
  ### Relevant Log Snippet
  `log
  D/MapsActivity: Preparing to fetch location reports for the last 24 hours!
  I/python.stdout: num_items is 2
  I/python.stdout: Fetching report for 48B8216B-4849-4CCF-BA75-877483CA1446 for the last 24 hours...
  I/python.stdout: Got 0 reports for 48B8216B-4849-4CCF-BA75-877483CA1446
  I/python.stdout: Fetching report for 050ACADF-DB12-47F1-A251-8B76EAEE88AE for the last 24 hours...
  I/python.stdout: Got 0 reports for 050ACADF-DB12-47F1-A251-8B76EAEE88AE
  D/MapsActivity: Did not get any location reports for beacon device with id
  48B8216B-4849-4CCF-BA75-877483CA1446. This device will not be shown in the UI.
  D/MapsActivity: Did not get any location reports for beacon device with id
  050ACADF-DB12-47F1-A251-8B76EAEE88AE. This device will not be shown in the UI.
  W/MapsActivity: Fond a beacon (48B8216B-4849-4CCF-BA75-877483CA1446) with no location history items! We
  can't draw such a beacon. Skipping...
  W/MapsActivity: Fond a beacon (050ACADF-DB12-47F1-A251-8B76EAEE88AE) with no location history items! We
  can't draw such a beacon. Skipping...
  I/MapsActivity: Finished visualising new location reports!

lequan2909, 2025-09-24:
> Hi [@lequan2909](https://github.com/lequan2909), thanks for the thorough report and investigation!
> 
> I think one more possibility is related to this: [#30](https://github.com/parawanderer/OpenTagViewer/issues/30)
> 
> This rotation mechanism in the current version of the FindMy.py library used in the app makes it so that some location updates end up not being locateable for tags that had their internal timer start to drift.
> 
> One thing you could do to deal with this for now is disconnect and reconnect the tags via your iOS device, which should reset any drift issue.
> 
> [That task](https://github.com/parawanderer/OpenTagViewer/issues/30) and subsequent fix (which is now pending on somebody else releasing a related tool, see that issue's page for details) might address this problem entirely

malmeloo, 2025-09-24:
I'm 90% certain that this is the known key drift issue, in which case the new version of the library will fix it. To test if that is indeed the problem you can re-pair your airtag and then re-export it, if it works then it's a key drift problem.

parawanderer, 2026-09-16:
Hi, this should now be [*mostly*](https://github.com/parawanderer/OpenTagViewer/issues/229) fixed as of [the new 1.1.0 release](https://github.com/parawanderer/OpenTagViewer/releases/tag/android-app-v1.1.0). Have a look, and feel free to open a new issue if you find more bugs.

closed 2026-09-16 by parawanderer